MillenniumMachines/Milo-v1.5 #57: M5x8 bolts on underside of table brackets are a bit short

dmason1992, 2024-02-25:
**Describe the bug**
M5x8 bolts on underside of table brackets are a bit short which means they barely engage in the t nuts

**Part Name / File Name**
X table brackets

**Required Resolution**
to avoid BOM changes suggest counterbore depth increased by 2mm to allow more threads to engage

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

yomgui1, 2024-07-11:
I've also noticed that.
On LDO parts they have added a countersink, not a counterbore. But it may not work. I've good printed pieces with perfect size tolerance and surface finish, my sourced screws have a flat surface, but even with that it's barely enough to engage the screw and have few threads.

I've also tested the M5x10 on original parts: may also not be a good solution for some people due to final tolerances: it works for me, but only a space of 0.2mm remains between the tip of the screw and the extrusion at a good tighten. That's too small.

I'm agree with you on adding a counterbore (than a countersink) is a good compromise to keep the BOM. But it may also decrease the strength of the parts... and these are structural parts.
So, I think 1mm of counterbor is enough to keep enough plastic and strength.

avnc, 2026-11-13:
No CAD changes are planned for 1.5.

closed 2026-11-13 by avnc