MillenniumMachines/Milo-v1.5 #58: TPU feet hole for insert size is too big

dmason1992, 2024-02-27:
**Describe the bug**
TPU feet hole for inserts is too big and they are loose

**Part Name / File Name**
TPU feet

**Required Resolution**
Reduce hole size (at least inline with other parts. possibly smaller due to flexy nature of TPU)

**Screenshots**
see stream at 54 minutes

(https://www.youtube.com/live/WmmRMIzL28I?si=z1HBU6jeW8mAj3Mf)

**Additional context**
Add any other context about the problem here.

ShaneDelmore, 2024-03-23:
[Milo_Foot_FIXED.stl.zip](https://github.com/MillenniumMachines/Milo-v1.5/files/14730388/Milo_Foot_FIXED.stl.zip)

Here is a fix, I tested it and the sizing was correct for me.  I am just uploading this, the fix was created by West3d.com.

avnc, 2026-11-13:
No CAD changes are planned for 1.5. The fixed foot STL from West3D in [this comment](https://github.com/MillenniumMachines/Milo-v1.5/issues/58#issuecomment-2016366954) solves it.

closed 2026-11-13 by avnc