MillenniumMachines/Milo-v1.5 #59: Bootstrapping in aluminum

tkircher, 2024-02-28:
Are there any plans to design parts for a more capable mill made from aluminum parts that can be machined on this mill? It's definitely possible for a conventional mill made from aluminum to machine steel.

EdwardChamberlain, 2024-02-28:
A lot of the parts on v1.5 could be adapted to be made from aluminium with minor modifications! currently we already recommend aluminium XY plates, FMJ, 'FIXATED' aluminium table and 'hearts of iron' metal spindle mounts.  It is on our radar and hopefully we can bring more bootstrap components to milo soon.

closed 2024-02-28 by EdwardChamberlain

tkircher, 2024-02-28:
The design requirements for 3D printing and for machining in aluminum are extremely different, given that aluminum parts don't have the anisotropy or material weaknesses of printed parts, but are also more limited to conventional machining steps. So a machine intended to be machined on this machine would have to be designed from scratch with that in mind, and would have a more conventional small-scale mill design, like dovetails and a saddle.