MillenniumMachines/Milo-v1.5 #60: Single piece skirts

dmason1992, 2024-03-06:
**Is your feature request related to a problem? Please describe.**
Single piece skirt would be stronger for the table than the two piece one. 

**Describe the solution you'd like**
Add option of single piece skirt for those with large enough printers.

**Describe alternatives you've considered**
Some form of mechanical joint between the two half width skirts to add the same support

**Additional context**
N/A

stvptrsn, 2024-03-08:
Corrected the heatset depth holes for the M5 inserts that go in the bolt down brackets to fit the LDO supplied inserts.
Changed from sacrificial bridges to "cross bridge" on the bolt down bracket M5 bolt holes.
Cleaned up some fillets on the rear skirt.
Combined front and rear skirts into one piece each.
[Milo_Skirts_1piece.zip](https://github.com/MillenniumMachines/Milo-v1.5/files/14532987/Milo_Skirts_1piece.zip)

stvptrsn, 2024-03-13:
Adding LDO modified one piece rear skirt.  Corrected power inlet tab locations.
[Rear_Skirt_LDO_1pc.zip](https://github.com/MillenniumMachines/Milo-v1.5/files/14594831/Rear_Skirt_LDO_1pc.zip)

avnc, 2026-10-27:
Closed by PR #61

commit 97272d81 (2024-03-08): Added 1 pc skirts. fixes #60
closed 2026-10-27 by avnc