MillenniumMachines/Milo-v1.5 #62: M3 heat set inserts are tight and  can melt through wall

EdwardChamberlain, 2024-03-08:
**Describe the bug**
Inserting the M3 heat set inserts in not the anti backlash nuts can cause the plastic to melt and block the space where the nut needs to go 

**Part Name / File Name**
X, Y, Z anti backlash nuts

**Required Resolution**
Open holes to reduce plastic movement

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.

mo22de, 2024-08-13:
I dont think you can make the M3 holes any bigger, and also you cant move them outwartds, because the backlash nuts have the holes like it is. If you reallly want to make this a better design, and i think that would make sence, cause ths is really not an optimal soution, we need other backlash nuts with a wider outer ring, or maybe a rectangular design, so the M3 holes can move outwards more.

I got only one wall line between the lead-screw hole and the heat-insert hole, printing it the same orientation like the picture shows. That didnt work out well. Walls were not even printed really bulding a wall.
So, I printed it again oriented with the lead srew hole showing up. The print turns out better and the wall between the holes is much better.

Still while pushing in the M3 heat-inserts, the inner hole can deform a little. I got it fitting by heating both heat inserts up, to temperature and than push in the lead screw nuts while ASA is still hot.

avnc, 2026-11-13:
No CAD changes are planned for 1.5. The anti-backlash nut was redesigned for Milo 2.0.

closed 2026-11-13 by avnc